adafruit/Adafruit_TouchScreen #20: Unknown RwReg for ESP8266

fim13, 2020-03-26:
Hi, 

I'm working with a ESP8266 and Adafruit SSD1306. Seems quite fare away I know. I include the library with platformio, so dependencies are loaded automatically.
```
Dependency Graph
|-- <Adafruit SSD1306> 2.2.1
|   |-- <Adafruit GFX Library> 1.7.5
|   |   |-- <Adafruit ILI9341> 1.5.4
|   |   |   |-- <Adafruit STMPE610> 1.1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0
|   |   |   |-- <Adafruit TouchScreen> 1.0.4
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- ...
```

Adafruit TouchScreen version 1.0.4 does not define `RwReg` if the board is unknown:
```
Adafruit TouchScreen_ID5430/TouchScreen.h:51:12: error: 'RwReg' does not name a type
   volatile RwReg *xp_port, *yp_port, *xm_port, *ym_port;
```
so the code does not compile.

I'll push to possible solution/work arounds.
- `typedef volatile uint32_t RwReg;` for ESP8266 as well
- or have an `#else typedef ...  RwReg;`

agrath, 2020-03-28:
The PR #22 didn't define for ESP8266

https://github.com/adafruit/Adafruit_TouchScreen/pull/23

ladyada, 2020-04-01:
added by hand. note esp does not have 2 analog pins.

closed 2020-04-01 by ladyada