adafruit/Adafruit_TouchScreen #27: TouchScreen.h does not work with Arduino UNO WiFi rev2 (__AVR_ATmega4809__)

andrewjohnsen31, 2020-04-22:
This is a fairly simple issue that I think Could be easily fixed.
the defined for Rev2 is "__AVR_ATmega4809__" this in not included in current release.
I had purchased " 3.5" TFT 320X480 + TOUCHSCREEN BREAKOUT BOARD W/MICROSD SOCKET" and was following the setup instructions. Understanding that my rev2 board does not match the getting started instructions. 

I went back a forth a bit, where this release works perfect with Arduino UNO, but not with rev2
Where the define is "__AVR_ATmega328P__"
Just switching boards give compile errors.

I spent a lot of time on this, so adding this board to TouchScreen.h could save someone else some time and frustration. Time is better spent playing with a really cool product.

**Released Code**
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega32U4__) ||              \
    defined(TEENSYDUINO) || defined(__AVR_ATmega2560__)
typedef volatile uint8_t RwReg;
#elif defined(ARDUINO_STM32_FEATHER)
typedef volatile uint32 RwReg;
#elif defined(NRF52_SERIES) || defined(ESP32) || defined(ESP8266) ||           \
   defined(ARDUINO_ARCH_STM32)
typedef volatile uint32_t RwReg;
#else
typedef volatile uint32_t RwReg;
#endif

**Recommended Change**
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega32U4__) ||              \
    defined(TEENSYDUINO) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega4809__)
typedef volatile uint8_t RwReg;
#elif defined(ARDUINO_STM32_FEATHER)
typedef volatile uint32 RwReg;
#elif defined(NRF52_SERIES) || defined(ESP32) || defined(ESP8266) ||           \
   defined(ARDUINO_ARCH_STM32)
typedef volatile uint32_t RwReg;
#else
typedef volatile uint32_t RwReg;
#endif

ladyada, 2020-04-22:
hi please try it and submit a working PR :)

andrewjohnsen31, 2020-04-23:
My first time with a pull request, yikes.
Its is an interesting song and dance, I like it and can only get better.

Let me know if more is needed.

gluca, 2022-01-22:
Same issue with Mega 1280 (and i guess same solution will work)