adafruit/Adafruit_TouchScreen #36: gpio: GPIO can only be used as input mode

BuzzerBoy, 2022-07-08:
I keep receiving this error whenever I try the library:

E (467781) gpio: gpio_set_level(226): GPIO output gpio_num error
E (467850) gpio: GPIO can only be used as input mode

The number in () is an error # that continues to count up.
It seems like the library keeps asking a defined input pin to behave like a write pin.

Hardware: 
SparkFun ESP32 Thing Plus
Adafruit 3.2" TFT LCD with Touchscreen Breakout Board w/MicroSD Socket - ILI9341 

Arduino IDE Version: 1.8.13
Touchscreen Library Version:: 1.1.3

Code:

// Touch screen library with X Y and Z (pressure) readings as well
// as oversampling to avoid 'bouncing'
// This demo code returns raw readings, public domain

#include <stdint.h>
#include "TouchScreen.h"

// These are the pins for the shield!
#define YP A2  // must be an analog pin, use "An" notation!
#define XM A3  // must be an analog pin, use "An" notation!
#define YM 21   // can be a digital pin
#define XP 14   // can be a digital pin

#define MINPRESSURE 10
#define MAXPRESSURE 1000

// For better pressure precision, we need to know the resistance
// between X+ and X- Use any multimeter to read it
// For the one we're using, its 300 ohms across the X plate
TouchScreen ts = TouchScreen(XP, YP, XM, YM, 332);

void setup(void) {
  Serial.begin(9600);
}

void loop(void) {
  // a point object holds x y and z coordinates
  TSPoint p = ts.getPoint();
  
  // we have some minimum pressure we consider 'valid'
  // pressure of 0 means no pressing!
  if (p.z > MINPRESSURE && p.z < MAXPRESSURE) {
     Serial.print("X = "); Serial.print(p.x);
     Serial.print("\tY = "); Serial.print(p.y);
     Serial.print("\tPressure = "); Serial.println(p.z);
  }
}

Things I've tried:
- adding #define ESP32, but that didn't help.
- changing the defined pins

closed 2022-07-08 by BuzzerBoy

contractorwolf, 2022-12-08:
@BuzzerBoy you seemed to have closed this issue.  If you resolved it maybe your could let others know what the problem actually was and how you resolved it.  That will help with future developers/makers and might even help yourself in the future when you hit a similar problem.

cuvtixo, 2022-12-09:
@contractorwolf, I would guess that he didn't solve the problem and closed it out of frustration. If you would prefer people don't do that, perhaps tell them to  just unsubscribe if they tire of waiting for an answer, and not close the issue.