adafruit/Adafruit_TouchScreen #39: Touch Screen library assumes 10 bit A/D?

professorwade, 2023-02-15:
Question: I noticed a number of 1024 and 1023 in the TouchScreen library and wonder if it is related to an assumption of a 10 bit A/D (i.e. Arduino 8-bit)? I am using the library with a SAMD51 part (Grand Central) and notice that the values tend to be much higher on the pressure.

https://github.com/adafruit/Adafruit_TouchScreen/blob/28e6e744982825d679f543b70480d31e457363b3/TouchScreen.cpp#L155

ladyada, 2023-05-03:
probably best to just set the adc resolution in the examples before use - wanna try that?
https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogreadresolution/

MaitrePoulier, 2023-05-03:
The problem with analogreadresolution() is that is work only with the: Zero, Due & MKR Family, not the ESP32-S2

ScruffR, 2023-10-15:
I just filed a pull-request to solve that problem
https://github.com/adafruit/Adafruit_TouchScreen/pull/43

TheTinyBang, 2024-01-04:
> I just filed a pull-request to solve that problem #43

Hi ScruffR,

thanks for your update, now it works with my ESP32.
I just have one issue left.
With an 10-Bit chip I can read out the whole touchscreen.
The MIN/MAX value of the Y axis goes (with Arduino MEGA 2560) from 110 to about 900.

By using your solution I can get the values from 0 to ~2200.
But the behaviour is really weird in this instance.
In the top left corner the value will read about 2200, while in the top right corner it is "only" about 1800.
In the bottom row I can see kinda the same. 
Bottom left the value will read about 300 and bottom right just about 0. Also in the bottom right corner there seems now to be dead zone where I cannot read any values anymore.

It seems like the whole touchscreen got moved "upwards" on the top left side and the bottom right side cannot be read anymore.
Tried it with 3 different touchscreens and 3 different ESP32, all the same result.

Did anyone experience the same?

I was also reading a article from this website: https://www.electronicwings.com/esp32/4-wire-resistive-touch-screen-interfacing-with-esp32
The autor there is using some other mods of the pinmode to achieve the results. Maybe there is a need to somehow get the TRI-State mode also into the library?

TheTinyBang, 2024-01-04:
Alright tried the code with an Arduino Nano ESP32 and with this Chipset it works fine. Kinda weird.
The other ESP32 NodeMCUs did not like it but I will try other chipsets and see if the results vary if I can.