adafruit/Adafruit_TouchScreen #45: Adafruit 2050 Touchscreen not working with Arduino Giga R1

MattGlobal, 2024-01-10:
[image]

Hello Everyone,

I made a working sketch with an arduino Mega 2560 but the looks very slow because of the SPI speed.
So i decide to use an arduino Giga R1.
Everything is working great with the ILI9341_GIGA_n Library and the screen.
So i tried to make work the touchscreen with the Giga R1 but It doesn't work.

I tried to configure on all ADC pins, I tried many Digital Pins  instead the ones of the original sketch (breaktouchpaint for example), I also tried the recommended wiring of screen's datasheet but the touchscreen never worked.

It's always reading the same value on the card. I can't use it with the Giga R1.

I Paste a screenshot of the reading values on my card for examples.

``
#include <stdint.h>
#include "SPI.h"
//#include "ILI9341_GIGA_n.h"
// *************** Change to your Pin numbers ***************
#define TFT_DC 9
#define TFT_RST 8
#define TFT_CS 10

#include "TouchScreen.h"

#define YP A2  // must be an analog pin, use "An" notation!
#define XM A1  // must be an analog pin, use "An" notation!
#define YM A3   // can be a digital pin
#define XP A4  // can be a digital pin

// This is calibration data for the raw touch data to the screen coordinates
#define TS_MINX 110
#define TS_MINY 80
#define TS_MAXX 860
#define TS_MAXY 925

#define MINPRESSURE 30
#define MAXPRESSURE 1000

#define DEBUG_PIN 0

#define USE_FRAME_BUFFER 1

//ILI9341_GIGA_n tft(&SPI1, TFT_CS, TFT_DC, TFT_RST);

TouchScreen ts = TouchScreen(XP, YP, XM, YM, 247);

void setup() 
{
//tft.begin();
Serial.begin(9600); 
//tft.fillScreen(ILI9341_BLACK);
//tft.setRotation(1);
}

void loop() 
{

// a point object holds x y and z coordinates

TSPoint p = ts.getPoint();
/*
if (p.z < MINPRESSURE || p.z > MAXPRESSURE)   //si la pression est supérieure à la tempo minpressure et inférieure à maxpressure, retour de l'information avec l'appel à la fonction return
          {
            return;
          }
*/

// Scale from ~0->1000 to tft.width using the calibration #'s
p.x = map(p.x, TS_MINX, TS_MAXX, 0, 320);   //vient chercher la position de x avec la calibration de la valeur TS_MINX, TS_Maxx, la première valeur est 0 la dernière valeur possible est la largeur de l'écran défini par la fonciton tft.width().
p.y = map(p.y, TS_MINY, TS_MAXY, 0, 480);

Serial.print("X = "); Serial.print(p.x);
Serial.print("\tY = "); Serial.print(p.y);
Serial.print("\tPressure = "); Serial.println(p.z); 

}
``

here is my code.

there is no mistake on the wiring, i think the problem came a library...
Or a definition of pin on the giga R1 maybe...

Thanks.

MattGlobal, 2024-01-10:
With this sketch, I just tried to connect every pins one by one on the GND without anything connected, and the result is still the same, the card is always reading the same values.

dhalbert, 2024-01-10:
I have been debugging this with @MattGlobal in discord. I suspect that the `analogRead()` values are varying too much, causing the readings to be discarded as invalid. Why that is the case, I don't know.

MattGlobal, 2024-01-10:
Hello @dhalbert , yes it's me, someone told me to post here in case of...

Thanks for your response.

MattGlobal, 2024-01-11:
Increasing the number of samples and uncommenting the checking code of the touchscreen library doesn't change the results on the card, it's still reading the same value (like the photo on my first post).

I wonder if it's not a problem about definition pins on the Giga R1.

I was thinking this library could help me :

https://github.com/stevstrong/TouchScreen_STM32

what do you think about that ?

dhalbert, 2024-01-14:
I got the same hardware and have reproduced the problem. I also got it to work properly with a Grand Central M4. Next step is to look at the a`nalogRead()` values and see what is different.

dhalbert, 2024-01-15:
I tracked this down to what looks like inadequate pin resetting by `analogRead()` on this board, and opened an issue.
- https://github.com/arduino/ArduinoCore-mbed/issues/829